Zer0-bit/gaggiuino #1: Love where this project is going

spoetnik, 2021-09-03:
I love where this project is going!
Would it be possible to build this project based on a ESP8266? This would add WiFi an Bluetooth for later use.

Zer0-bit, 2021-09-03:
Hi Martijn, thanks for your interest in the project!
I don't see any reason why it wouldn't be possible to use a ESP8266 board coupled with all the other components already used here, there is a small chunk of code which will need to be changed as it's Nano specific, more specifically it's chip specific as it sets the state of the pins directly on the chip rather than on the controller but once that is changed i think it should work with that board no issues.

Zer0-bit, 2021-09-03:
> I love where this project is going!
> Would it be possible to build this project based on a ESP8266? This would add WiFi an Bluetooth for later use.

Hi Martijn, thanks for your interest in the project!
I don't see any reason why it wouldn't be possible to use a ESP8266 board coupled with all the other components already used here, there is a small chunk of code which will need to be changed as it's Nano specific, more specifically it's chip specific as it sets the state of the pins directly on the chip rather than on the controller but once that is changed i think it should work with that board no issues.

Zer0-bit, 2021-09-04:
I've added a new branch where i compiled it against the esp8266 module, it does verify successfully but obviously i have no means to check if it actually works, the pinout is defined according to what i found online are the pin numbers  in the arduino ide.

spoetnik, 2021-09-04:
Thanks!  I will try to upload the code to an esp8266 when I have the time. I don’t have all the components to check for the correct working, but I will order them soon form alie. 
thanks!

spoetnik, 2021-09-05:
I am new to all this. Entered this bug in the RBDDimmer library :: https://github.com/RobotDynOfficial/RBDDimmer/issues/14
The suggested fix works. The code compiles, end flashes the ESP8266

I wil order the rest of the components, so I can verify the working

Zer0-bit, 2021-09-05:
Great news, now you'll just play the waiting game as delivery might take until 1 month on some pieces but glad it uploaded  all fine.

closed 2021-09-16 by spoetnik

spoetnik, 2021-09-17:
Building a Bluetooth capable Gaggiuino could be a Decent Espresso DE machine killer.
https://www.home-barista.com/blog/decent-espresso-de1-review-t57610.html

The android app for this machine is opensource. How cool would it be to control a Gaggia with that app......

Zer0-bit, 2021-09-17:
That was my original idea as well then i remembered i have a kettle that offers kinda the same and i've not oppened that app in months now, in result i understood I don't like the concept of the DE approach, information given in a really inconvenient way, i mean all that info is gimmick and loses appeal 1 week later also having to pull the phone or a tablet constantly is kinda making the process less enjoyable so as a result i decided it's not worth the effort imho.