Zer0-bit/gaggiuino #48: Scale

poxin13, 2021-09-19:
Very interested in this project, thank you for posting it. I've got an auber pid on mine at the moment but the dimmer would be great!

Have you thought about adding bluetooth support to link up to scales like an acaia lunar? Looks like this would require a different arduino for that functionality but would allow to stop brew at a predefined ratio. Could be very cool.

Zer0-bit, 2021-09-20:
Didn't think about that but thought about integrating a  load cell in the water tray for achieving this result, i'm not sure i want to start integrating with other commercially available products in this sense, as then people will keep creating issues to add integration with <insert brand>, i will research it and definitely see what options are out there but for the time being no definitive plans in regards to that.

poxin13, 2021-09-20:
100% understand not wanting to integrate with a specific make/model - was just using the lunar as a reference because it's the most popular one. Basically was borrowing the idea from https://decentespresso.com/overview for bluetooth support.

> This allows your espresso to stop at a desired weight. You also can see the realtime flow rate of coffee into your cup. Atomax Skale 2, Decent Scale, Acaia Pearl, Acaia Lunar, Felicita arc, Hiroia Jimmy.

Will likely try putting together this project in the future hopefully I can help expand on it.

closed 2021-09-20 by poxin13