Zer0-bit/gaggiuino #61: Connecting the relay

spoetnik, 2021-09-25:
**Describe the bug**
After connecting all components, I think my relay isn’t switching. There is no led indicating change. 
I have heated up my thermocouple from way below the set temperature, to above the set temperature, and I see no led indicating change. 

The pinout in the readme states the ground should be connected to relay pin nr. 3. Is this correct? Shouldn’t this be the 5 volt?

Zer0-bit, 2021-09-26:
Yes good spot i had mistakenly specified the pins wrongly, check again it's corrected now.

spoetnik, 2021-09-26:
Thanks, the pins are correct now in the README. Hoping to check my "Hobby Gaggia" next week, so I can build in the upgrade.
I am using the nano for now. You have convinced me about the simplicity of the project using the nano, And I will not use a esp8266 for now. Thanks for making the branch.

closed 2021-09-26 by spoetnik