Zer0-bit/gaggiuino #65: Documentation Update

matate99, 2021-09-27:
Not a big git person, but I did mark up the Classic Pro wiring schematic to make it a little more clear what goes where. Not sure if this is something you want to include in the instruction
[GAGGIA_CLASSIC_2019_Wiring_Schematic.pdf](https://github.com/Zer0-bit/gaggiuino/files/7237419/GAGGIA_CLASSIC_2019_Wiring_Schematic.pdf)
s.

Zer0-bit, 2021-09-27:
Oh dang! Awesome dude! I was literally struggling to come up with this in a comprehensible way, this is just amazing.
So you can fork the **[dev]** branch into your account add the schematics to **schematics/<filename>** when you upload a file,  if you add a path style name as the name of the file on upload it'll create the folders if they don't exist.
Then after it all looks nice open a pull request to merge back into the dev branch of this account, this way your name will appear as a contributor.

matate99, 2021-09-27:
I couldn't find a way to edit the name of the file to get it in a folder, but I think it's there now. Again, very new to git so sorry if you need to walk me through. Also, thanks for everything you've done on this. Way more fun than just buying a ready-made PID :D

Zer0-bit, 2021-09-27:
No worries, git is nice but it requires getting used to if it's not smth you use on a daily.
In regards to the mod it's not even yet complete, features are still in plans so it'll just become way more feature complete i'm currently waiting for a new batch of test equipment so fun times aren't over ))
It might not be for everyone but definitely who puts in the effort is greatly rewarded.

closed 2021-09-27 by Zer0-bit