Zer0-bit/gaggiuino #79: Best place to leave the USB charger?

dlime, 2021-10-04:
I'm undecided on where to leave the USB charger for Arduino. 
Whether inside/outside the back of the machine :)

Outside:
- look uglier
- practical for Arduino code updates
- more safe in terms of water leaks / contact 

Inside:
- neat look
- risk of getting wet
- less practical for updating Arduino code

Did you already experiment with the different possibilities?
Where did you left yours?

Zer0-bit, 2021-10-04:
> I'm undecided on where to leave the USB charger for Arduino. Whether inside/outside the back of the machine :)
>
> Outside:
>
> * look uglier
> * practical for Arduino code updates
> * more safe in terms of [...]

Please don't create github issues about this type of stuff, there's a discord server created to discuss installation troubles and general ideas, issues are mostly for code issues or code improvement ideas.

closed 2021-10-04 by Zer0-bit

Zer0-bit, 2021-10-04:
Btw the answer you're looking for has been answered and showed in pictures on the discord server, as LikeableBump has posted pictures on how he integrated all with db15 connectors.

dlime, 2021-10-05:
> Please don't create github issues about this type of stuff, there's a discord server created to discuss installation troubles and general ideas, issues are mostly for code issues or code improvement [...]

All right see you on Discord. Decided to ask it here since up till now I've seen both HW/SW questions in Issues 🤷

Zer0-bit, 2021-10-05:
I wasn't trying to come up arrogant btw, i think it's a issue with trying to express yourself over text on the internet lol