Zer0-bit/gaggiuino #170: Features + overview in the README

jbohnslav, 2022-01-21:
**Is your feature request related to a problem? Please describe.**
The README does a great job at describing how to build the Gaggiuino. However, the description and feature list is incredibly sparse. I'm not 100% sure what this project would let me do or not do. The videos are a start, but they're not a replacement for documentation.

**Describe the solution you'd like**

## overview
First, a nice OVERVIEW at the top of the README. For example, "this mod lets you control the temperature and pressure profiles of your Gaggia. It allows you to preinfuse your shots. Future features include a built-in scale that lets you graph your shot over time and even automatically stop a shot (will it?). The overall parts list costs around $100 USD (or whatever) if you buy in the United States, or XXX currency in YYY country. However, pre-built machines with similar amounts of control cost many thousands of dollars. You should buy the parts ASAP, because they are back-ordered."

## An extended "Features / Uses" section in the README

Each of the bullet points in the "Features / Completeness" list could be its own subsection of the README. I'm not sure what exactly all of these features are, or why I would want them necessarily. 

-  `Temp control:` Like the PID mods, this project improves the infamously bad temperature control of the Gaggia. We use a simple pulse-width modulation temperature control system to control temperature. You can set whatever temperature you want (can you?) etc. etc. 
-  `Steam control and alerting:` what does this mean?
-  `Preinfusion:` does it just allow for 5 second 1-bar preinfusion? can you do profiles? I'm not sure!
- ` Realtime pressure feedback: `does this require the "pressure sensor" in the EXTENDED FUNCTIONALITY section of the assembly instructions?
-  `Pressure profiling (realtime feedback loop)`
-  `Manual pressure control`: does this require the dimmer module?
-  `Realtime values update`: what values?
-  `Saving selected values: ` what selected values?
-  `Descale program`
-  `Brew start sense`: what does this mean?
-  `Auto shot timer:` what does this mean?

## A description of the "Extended functionality" parts
What features do I need these parts for?

**TL;DR: documentation!**

VR29, 2022-01-21:
I think most of your questions are quite obvious. As this is an open-source project, you could always further investigate and improve the documentation yourself.

shigadeyo, 2022-01-22:
I have all of the parts on the way to build this mod and planned to enhance the documentation along the way as part of my contribution to this project; please feel free to check back in a a month or two once the components have arrived and I have had a change to update the documentation and submit it for review

Zer0-bit, 2022-01-24:
Not an issue.

closed 2022-01-24 by Zer0-bit