Zer0-bit/gaggiuino #171: Choosing an Arduino variant to order

joshcomley, 2022-01-23:
On the [link provided](https://www.aliexpress.com/item/4000903444456.html?spm=a2g0s.9042311.0.0.1b8c4c4dzBsUSa) there are some variations, and in all other hardware listings the variant is very handily specified.

So I'm wondering does it matter which of these I order? Can I order the USB-C one without issue?

[image]

Thanks for this project!

shigadeyo, 2022-01-23:
> On the [link provided](https://www.aliexpress.com/item/4000903444456.html?spm=a2g0s.9042311.0.0.1b8c4c4dzBsUSa) there are some variations, and in all other hardware listings the variant is very [...]
My understanding is yes and I'm pretty sure that Zer0-bit stated in the Discord that he uses the USB-C one. I would have done the same, but found out later that the the only real difference in them is the connector option.

shigadeyo, 2022-01-23:
Found it! Here is the link to the comment about this in the Discord channel - https://discord.com/channels/890339612441063494/890339612441063497/895691100201844806

Zer0-bit, 2022-01-24:
Not an issue.

closed 2022-01-24 by Zer0-bit