Zer0-bit/gaggiuino #172: Variable Screen Size Files

kwood94xj, 2022-01-26:
I have all the items on order that you have listed but I mistakenly got the 3.5" screen. I have loaded up everything in HMI and found that the software wont scale the content to the new screen size. What would be the best way to scale this? Could you share your PNG files?

shigadeyo, 2022-01-26:
I believe that the feedback you will receive is that only the 2.4" screen is officially supported at this time; there may be additional help available in the Discord channel, but 2.4" is the stated requirement

kwood94xj, 2022-01-26:
I am willing to put in the work to scale everything up if the original PNG files are available.

Zer0-bit, 2022-01-27:
As @shigadeyo stated.
Closing issue as it's not an issue.

closed 2022-01-27 by Zer0-bit