Zer0-bit/gaggiuino #175: Question about the Gaggia Classic Pro Steam Control wiring

LindenLiu, 2022-02-11:
**Is your feature request related to a problem? Please describe.**
On dev branch, I see your update on the Steam control wiring which connect gnd and a7 to the switch. That's smart.  But I think there is an issue for the steps:

1. Move steam switch wire 4 to steam switch pole 1.
2. Unplug and secure steam switch wire 5.
3. Replace the SSR brew thermostat connected wires with the steam thermostat wires.
4. Bridge the brew thermostat wires(the wires we just disconnected from te SSR in favour of the steam thermostat ones) 

**Describe the solution you'd like**
1. I think there is no need to move steam switch wire 4 to switch pole 1. It can be disconnected directly. Instead move the wire 4 to pole 1. The LED light white wire which connect to wire 4 could be disconnected from LED and have a wire connect the LED light to switch pole 1. My thought is that, when steam switch is on, we could set boiler (brew) SSR to HIGH to make the boiler keep heating until the desired steam temperature reached. And leave the steam thermostat as a safety fallback. The steam thermostate max at 145 C, as long as the steam temperature is not beyond 145 C, it will allow the boiler keep heating if the brew SSR is high. 
2. Steam thermostat wires should remain there.

My understanding on Gaggia Wire Diagram, the steam thermostat and brew thermostat is kind of in serials. The steam thermostat's output connect directly to boiler, and it has two input source: one is steam switcher, another is the output of brew thermostat. So when steam switch is OFF, the power came from the brew thermostat if it is not exceeding the brew temperature. If the temperature exceeds the brew temperature, then the brew thermostat shut off the power. If steam switch is ON, then the brew thermostat will be shortcut by the steam switch and only steam thermostat in effect. 

So with the arduino controlling the brew thermostat (brew SSR), there is no need to have a second power source from steam switch to the steam thermostat and only need to connect the LED to the steam switch to keep the light indication working. 

Not a electrician, would love to hear your feedback on that.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

LindenLiu, 2022-02-11:
[image]

This is the wiring in my mind. I would like to keep the steam thermostat untouched as safe fallback to cap the temperature at 145 c maximum

Zer0-bit, 2022-02-11:
I'm not sure this will work, if you have a gcp go with your design see what results you get.
I'm going to close this as it's most certainly not a issue, the steam handling does work in it's current form.

closed 2022-02-11 by Zer0-bit