Zer0-bit/gaggiuino #252: Wiki: Home.md markdown and formatting edit

EdwardAngert, 2022-07-12:
Hi friends!
I started making some markdown and formatting edits to the wiki, but there's no way to PR to Wikis on GitHub¹ 😞

My goal is to make both the markdown and the rendered page easier to parse and make edits to later.

So, check out what my suggestion looks like on my fork: [EdwardAngert/gaggiuino/wiki](https://github.com/EdwardAngert/gaggiuino/wiki)
- here's the commit: [5c07a29](https://github.com/EdwardAngert/gaggiuino/wiki/Home/5c07a297433475c3f8839a121f67252da9c02a8f)
- here's the markdown if you like it and want to paste it in: [Home.md](https://raw.githubusercontent.com/wiki/EdwardAngert/gaggiuino/Home.md) (I'm also happy to make the edits if you're open to it)

I hope to be able to make more edits while I work out whether this is a modification I can take on. I just like formatting markdown in my spare time while I read instructions 🤷 

---
¹ There is a way to [enable public editing](https://docs.github.com/en/communities/documenting-your-project-with-wikis/changing-access-permissions-for-wikis) though. Under the [repository settings](https://github.com/Zer0-bit/gaggiuino/settings#features), uncheck the box next to `Restrict editing to collaborators only`.

Zer0-bit, 2022-07-12:
Hello - welp aren't you on time with the issue, wiki is moving to it's own repo so it can be managed with proper docs managing software.
Repo: [Zer0-bit/zer0-bit.github.io](https://github.com/Zer0-bit/zer0-bit.github.io)
Docs managing software: [Docslify](https://docsify.js.org/#/quickstart)

You can ofc just fork and raise PRs in that repo as normal on any git platform.

EdwardAngert, 2022-07-12:
beautiful!
congratulations!

I'll close this issue and get a PR going to the docs repo

closed 2022-07-12 by EdwardAngert