Zer0-bit/gaggiuino #418: Predictive Weight Start Inaccurate in Flow Mode

kokonstas, 2023-02-28:
When in flow control, predictive weight starts far too early. 
When in pressure control, I'm within +/-1 gram. I assumed this to be a machine/setup specific issue, but I'm unsure why the various checks linked below would only need to be applied to Pressure Control. Removing the if statement that checks the mode makes the predictive scale work uniformly for me.
https://github.com/Zer0-bit/gaggiuino/blob/400ba137955e12bf0cd7f8bead86368d516d4b63/src/functional/predictive_weight.h#L54

Zer0-bit, 2023-03-01:
Hi, thx for your input but predictive weight is a very complex subject and it's behaviour can't really be discussed in such a format.
My changes are based on observations across multiple weeks of testing i've been doing on my end and i find it's behaviour quite accurate.
This said you can always tare manually while on the brew screen by clicking the circular arrow button to reset the predictive/weighing mechanism for the situation you're trying to follow.

kokonstas, 2023-03-01:
If checking for pressure to cross 1.1 bars on Flow Control is working reliably for you and everyone else (which is pretty evident in the predictive scales channel), I'll assume this is a personal issue.

Thanks for looking at.