Zer0-bit/gaggiuino #471: [Web Interface] TypeScript?

ckifer, 2023-06-21:
React and web development standards have been moving towards TypeScript as a rule for a few years now. JavaScript only projects tend to lead to issues down the road, especially when you have consumers (even if the consumers are just deploying/running their own copy).

Strongly typed languages (even in the web world) help with a multitude of things - developer experience, better autocomplete, assurance in what you are getting when variables are accessed, the same result at the end of the day, etc.

Any reasons why the web-interface isn't using TS? If not would a refactor be welcomed?

Zer0-bit, 2023-06-21:
No reason tbh, first time touching this shit so it's basically learning this as i go, to me personally there's basically no difference which language to choose since gonna need to get a grip on either anyway.

kstam, 2023-06-22:
No issues with a refactor either if you can be bothered :) I like Typescript as a language. Just couldn't be bothered to do all the type definitions when I was starting it.

ckifer, 2023-06-22:
Cool 

I started moving things to vite (https://vitejs.dev/) because this ejected Create React App setup will become relatively unmanageable over time (large, unreadable config files, unneeded dependencies, etc) and it is no longer recommended by the React team. 

Let me know if thats over-stepping, but essentially we end up with the same thing as before + TypeScript support/TypeScript first + faster dev server (vite has much better HMR) + easier configuration structure

results in the same proxy, same functionality, less bloat
[image]
[image]

Zer0-bit, 2023-06-26:
Issue linked to an already merged PR, closing.

closed 2023-06-26 by Zer0-bit