Zer0-bit/gaggiuino #544: CONTRIBUTING.md?

markandersontrocme, 2024-01-22:
**Is your feature request related to a problem? Please describe.**
Any docs on how to contribute to the project? I am interested in working a bit on the webserver and was wondering how contributions work

**Describe the solution you'd like**
a CONTRIBUTING.md file that contains how contributions work.

**Describe alternatives you've considered**
I guess a section in the README.md...

**Additional context**
Add any other context or screenshots about the feature request here.

markandersontrocme, 2024-01-22:
@JackPoint thanks, I'm already in the discord. Is there a channel for contributions? I read in another issue/pr that zer0-bit said there was work being done in a private repo. So maybe only a subset of people are working/contributing to the code?

JackPoint, 2024-01-22:
As far as I understand, the current development is stopped (for stm32+nextion). There is a private repo where the new development takes place.

chwzr, 2024-01-30:
would also love to contribute to frontend