Zer0-bit/gaggiuino #561: After the OTA update, the screen displays a blank screen.

szongr, 2025-02-16:
[image]
[image]

craigywsm, 2025-02-17:
@szongr did you flash the right file for your board.
Ref: https://gaggiuino.github.io/#/guides-stm32/mcu-flashing
I flashed the performance to my V3.0 board
once flashing the PCB-***.bin it started to work after a reboot

Zer0-bit, 2025-02-22:
The white screen is expected, it relates to an unannounced hardware change the manufacturer of these lcds made, this will be fixed soon.

Zer0-bit, 2025-02-23:
you can now upgrade to the latest release which fixes this

closed 2025-02-23 by Zer0-bit

gregbergevin, 2025-02-23:
Should be resolved with latest release:
https://github.com/Zer0-bit/gaggiuino/releases/tag/dev-6655d6d