Zer0-bit/gaggiuino #562: Not able to update core file

craigywsm, 2025-02-17:
I found I was not able to update the core version using the web interface
It uploads ok but does not upgrade
This was the same as the previous version also 

[image]

Zer0-bit, 2025-02-22:
https://gaggiuino.github.io/#/guides-stm32/mcu-flashing?id=ota-esp32-stm32u585

craigywsm, 2025-02-22:
@Zer0-bit yes is did try updating the core first with the lease a few weeks ago and this failed. 
I notice in the release notes for the latest release you mentioned to do the core after the ui updates. 
This also did not work. I needed to flash via stm

Zer0-bit, 2025-02-23:
Your core doesn't support to be updated via OTA as that's  a feature available only to the u585 based core.

closed 2025-02-23 by Zer0-bit

craigywsm, 2025-02-24:
Ok I was able to update the core from the 1st to the 2nd release.