Zer0-bit/gaggiuino #564: Can't update core firmware

daveramsaran, 2025-02-22:
Help core firmware can not be updated, tried reverting to previous firmware no luck. Core firmware is currently 0000

Zer0-bit, 2025-02-22:
If you didn't follow the advice to update the CORE first and do the other 2 UI binaries afterwards then you will have to flash the core firmware using stlink.

daveramsaran, 2025-02-22:
Ok, thank you. I initially started in order and may have second guess my self when the scale stop communicating and redid the flash out of order. Is there a tutorial on using STlink to flash? Or is it self explanatory

daveramsaran, 2025-02-23:
Thank you guys, was able to restore and update with a combination of stmlink cube and OTA. Think I'm getting the hang out it.
Steps I did:
*Flash previous core firmware with stmlink cube
*OTA with current firmware then immediately embedded and web before restarting unit to verify.

closed 2025-02-23 by Zer0-bit