Zer0-bit/gaggiuino #566: v.6655d6d

wangmoumou22434, 2025-02-24:
After the upgrade to v.6655d6d, the screen jitter and white screen issues have indeed been resolved. However, a new problem has emerged. When the Bluetooth is turned on to connect to the Eclair electronic scale, the screen will restart frequently and repeatedly.**Describe the bug**
A clear and concise description of what the bug is.

Zer0-bit, 2025-02-24:
Can't replicate, used bookoo to pull shots, left it connected until timeout and had no issues, might be something to do with the eclair implementation hard to say since i cna only test using what i have.

Zer0-bit, 2025-02-24:
To exclude any build issues i re-built the binary and uploaded it again to the same release, cna you download the new one again and test ?

wangmoumou22434, 2025-02-24:
> 为了排除任何构建问题，我重新构建了二进制文件并将其再次上传到同一版本，您能否再次下载新版本并进行测试？

could give it a try

closed 2025-02-24 by wangmoumou22434