Zer0-bit/gaggiuino #567: Core update failed

Sursur123, 2025-02-24:
I followed the instructions on the release page and first upgraded to 616ea70. After that I started with embedded and then web - which oth worked.
Then when upgrading the Core it got stuck at 24%.

[image]

LCD does not flicker anymore but machine cannot be used now anymore.
Web reads following:

[image]

I guess I have to fix it with STlink?

Thanks!

Zer0-bit, 2025-02-24:
Yeah stlink will fix this, i'll test later on my end if i can see the same behaviour as maybe in my chase to fix the flicker/white screen i optimised smth too much lol

Sursur123, 2025-02-24:
I was able to fix it with STlink.
Core now reads 

[image]
 without "a" at the end...

But machine works again - thats most important :)

closed 2025-02-24 by Sursur123