Zer0-bit/gaggiuino #568: v.6655d6d Upload again. I'm not sure how to get in touch, so I choose to upload again.

wangmoumou22434, 2025-02-24:
I downloaded and updated the file uploaded 17 minutes ago. After trying to connect to the Eclair electronic scale, the screen still restarts repeatedly, and the previous problem has not been resolved.

https://github.com/user-attachments/assets/06d3c8d4-3f56-4dca-87bf-53bdd25aec26
https://github.com/user-attachments/assets/97c0f4b7-b69a-4937-85cc-d96799c2d6af

AtomHeart-Lang, 2025-02-24:
The crash issue when connecting to Eclair also happened on another machine. Both devices worked fine with the Jan 6th firmware before, and even after rolling back from the latest update. The only sus part I noticed is this code tweak related to the NimBLE-Arduino here: https://github.com/Zer0-bit/esp-arduino-ble-scales/commit/0075c761d603af8cfddaf4c8cf29a573b36172f8 Honestly, I can’t say for sure if the issue is right here . Plus, I can’t see any other changes tied to the connection stuff when comparing the latest firmware with the Jan 6th one.

wangmoumou22434, 2025-02-26:
[image]
I'm not sure if the version of February 26, 2025 is the latest. After I upgraded, the device still keeps restarting repeatedly when connecting to Bluetooth. It seems that the problem hasn't been solved.

Zer0-bit, 2025-02-27:
Update again pls, i updated the ui-embedded.bin a few minutes ago again. let's see if this fixed the issue.

wangmoumou22434, 2025-02-27:
> 请再次更新，几分钟前我又更新了ui-embedded.bin。让我们看看这是否解决了问题。

There's been some progress. After making the connection, the screen no longer keeps restarting. However, no data is shown on the screen. I've recorded a video.

wangmoumou22434, 2025-02-27:
> 请再次更新，几分钟前我又更新了ui-embedded.bin。让我们看看这是否解决了问题。

After the screen is connected to the host, it is said that it can be used normally. When the screen is powered separately, it can establish a connection, but there is no data transmission.

Zer0-bit, 2025-02-28:
Update again and see if it works now

wangmoumou22434, 2025-03-01:
Congratulations! It can be used normally.

closed 2025-03-01 by Zer0-bit