Zer0-bit/gaggiuino #585: sprofiler.io upload fails

danho-san, 2025-07-02:
**Describe the bug**
Got an API token for my machine on sprofiler.io, addid it into the respective credentials field under Experimental.
When selecting Upload in a shot history, I get "Failed to upload shot to all services."

**To Reproduce**
Steps to reproduce the behavior:
See above

**Expected behavior**
Upload success.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: WIN11
 - Browser: Edge
 - Version 138.0.3351.55 (Offizielles Build) (64-Bit)

**Smartphone (please complete the following information):**
./.

**Additional context**
Core:1970caf
Mbed:1970cafe
Web:6655d6da

Zer0-bit, 2025-07-06:
You need to update Web to latest, also if you're using a BT scale it's a known bug upload fails with them, bug is fixed in latest beta and will cone in the coming release.

Zer0-bit, 2025-07-18:
Any new input on the issue ?

closed 2025-07-21 by Zer0-bit