Zer0-bit/gaggiuino #591: Not able to use profiles with id > 15

kpp39, 2025-08-08:
It seems that every time when you create new profile or duplicate existing one, there are an id, that incremented no matter how many profiles are presented in machine. And the bug is when id > 15, you cannot save new profiles or even save changes in existing ones, it broke something somehow and there are no way to fix that or "factory reset" the machine

[image]

kpp39, 2025-08-08:
And i can delete this profiles via REST API, but i can't create new profiles (because id for that profiles still > 15 even when i have only five profiles) and even worse, i can't save modifications to existing ones

Zer0-bit, 2025-08-12:
[image]
This isn't a issue for everyone else, seems to be very specific to whatever happened on your end.

closed 2025-08-22 by Zer0-bit